OPCF-Members/UA-.NET-Legacy #1: Handle leak in Tcp/TcpAsyncOperation.cs and Transport/AsyncResultBase.cs

dk2129, 2015-10-29:
TcpAsyncOperation implements IDisposable, but Dispose is never called.
The same for AsyncResultBase.
The TcpAsyncOperation.End method initializes an Event and uses it to wait the operation completion. However, this event cannot be freed determinstically since the opertion are never disposed.
A simple correction is to free the event just after the operation completion in the End method:

 // wait for completion.
            if (mustWait)
            {
                if (!m_event.WaitOne(timeout, false))
                {
// Added :
                    m_event.Close();
                    m_event = null;
                    throw new ServiceResultException(StatusCodes.BadRequestInterrupted);
                }
// Added :
                m_event.Close();
                m_event = null;
            }

However the AsyncWaitHandle property should never be called, since we do not know if the event will be destroyed or not. Hopefully, this property is dead code and never called.

Same with AsyncResultBase.WaitForComplete and AsyncWaitHandle.

STEPS TO REPRODUCE:
36h certification test, with 5 clients, 1000 MonitoredItems/client, nodes changing at rate 1000 times/200ms
Small handle leak.

AlinMoldovean, 2016-03-02:
Fixed in the 552f5c2d8eed8df3eb34d1d00edafa82884e2707 commit.

closed 2016-03-02 by AlinMoldovean

AlinMoldovean, 2018-06-18:
Hello @BrunoJuchli ,

Thank you for pointing this out.
The issue does not apply for UA .NET Standard stack because that part was reworked.
We also put a fix to avoid the race condition, so it should be ok also on this repository.