OPCF-Members/UA-.NET-Legacy #4: When OPC UA server Registers with LDS will always recieve a BadSecureChannelClosed exception

nathanpocock, 2015-10-30:
A UA server derived from StandardServer in the SDK using the .NET Framework Stack seems to always get a BadSecureChannelClosed exception when registering with the LDS. Note that the Registration itself succeeds, just that it seems to be abruptly terminating the SecureChannel which the .NET Framework Stack does not like. This may be by design, although seems kind of rough.

The actual issue is initiated in the TcpMessageSoccket.GetNextMessage method where the messageChunk.Array is null.

Because it appears that the Certificates between the LDS and the Server are mutually trusted, and the registration works properly I have to assume this is something everyone is just ignoring in the exception handling. Of course I may have something configured incorrectly... 

--

Sample UA Server should exhibit the same behavior.

MatKubicki, 2015-11-16:
I believe I am seeing this as well, I get the same error being thrown within TcpAsyncOperation.End(), on digging the serviceresult object is coming from within TcpMessgeSocket.GetNextMessage (as you mention).  Again the discovery registration works but this is very annoying when you are in a debug session as this error occurs every 30 seconds or so.

AlinMoldovean, 2016-03-04:
BadSecureChannelClosed is one of the accepted and expected errors when closing a channel. The same for BadRequestInterrupted.

	// wait for the close to succeed.
	if (operation != null)
	{
		try
		{
			operation.End(timeout);
		}
		catch (ServiceResultException e)
		{
			switch (e.StatusCode)
			{
				case StatusCodes.BadRequestInterrupted:
				case StatusCodes.BadSecureChannelClosed:
				{
					break;
				}

				default:
				{
					Utils.Trace(e, "Could not gracefully close the channel.");
					break;
				}
			}
		
The BadSecureChannelClosed for the operation is set by the TcpClientChannel.Shutdown() method:

	foreach (WriteOperation operation in operations)
	{
		operation.Fault(new ServiceResult(StatusCodes.BadSecureChannelClosed, reason));
	}

The fix in this case would be to supress the log entry for the BadSecureChannelClosed error, from the ServiceResultException constructor.

commit d2800369 (2016-03-04): Fixes #4 - BadSecureChannelClosed error in log file when registering with LDS.
closed 2016-03-04 by AlinMoldovean