OPCF-Members/UA-.NET-Legacy #6: How to detect that status and timestamp is writable (CTT also)

nathanpocock, 2015-10-30:
Currently there is no way to detect if a server supports writing of status and or timestamp.

One option would be to have a server wide capability information about general support of this feature.

A more specific information would be to extend the access level with additional flags for StatusWrite and TimestampWrite.

--

Discussed at IOP Nuremberg 2013
Additional information / discussion through Christian Hock / Karl Deiretsbacher

nathanpocock, 2015-10-30:
UA 1.03 adds provisions for this in the AccessLevel bitmask. This feature should be postponed in the stack until we launch a 1.03 version.  Will postpone until the 1.03 version of the stack is created.

AlinMoldovean, 2016-07-11:
The list of AccessLevels was extended with StatusWrite and TimestampWrite values.

closed 2016-07-11 by AlinMoldovean