OPCF-Members/UA-.NET-Legacy #7: Recreating the session does not honor original client handles of monitored items

milankase, 2015-10-30:
When the client session is recreated (the Session.Recreate method, used e.g. during reconnection) then monitored items in the new subscriptions of the newly created session does not honor original client handles, since during recreation the items are assigned autogenerated handles. Since the client handle is the only way to identify the item during session notification (or is it not?) this quite complicates things. One would legitimately expect that after reconnection (using the SessionReconnectHandler) the returned session is in the same state as the original one.

Is this a bug or a feature? How to identify items during session notification if not using client handles? What is the expected use case?

AlinMoldovean, 2015-11-03:
ClientHandle is assigned using an internal counter and is incremented every time a new MonitoredItem object is created, also during a Session.Recreate() procedure. 
The correct way of identifying the recreated monitored items is by setting the "Handle" property of the object at application level.
This reference will be reassigned when creating the new object from the initial template and can be used for identifying the newly created objects.

milankase, 2015-11-03:
Then I'm not sure how to get this Handle during session notification. The MonitoredItemNotification only contains ClientHandle, not the application level Handle. Should I not use the session notification for Monitored Items?

AlinMoldovean, 2015-11-04:
The MonitoredItem.Notification event can be used in this case. 
It also provides the MonitoredItem object:
	MonitoredItemNotificationEventHandler(MonitoredItem monitoredItem, MonitoredItemNotificationEventArgs e)

milankase, 2015-11-06:
Thanks, I didn't know about the Notification event.

closed 2015-11-06 by milankase