OPCF-Members/UA-.NET-Legacy #14: The CertificateGenerator utility is not installed

Warper2001, 2016-01-04:
I cloned and built the git repo. After building the repo I opened QuickStarts.DataAccessServer.exe just like the tutorial says but I received an exception since the certification could not be generated from the utility which doesn't seem to exist.  How can I get the CertificateGenerator utility? 
[image]

Thanks!

nathanpocock, 2016-01-04:
Hello, I would suggest downloading the sample applications installer for now, until this code is incorporated into the repository. You will need to login to the OPCF website; you can create a free account if your company is not a member: https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications/

Warper2001, 2016-01-05:
Thanks for the help Nathan, it's loading up fine now.

closed 2016-01-06 by Warper2001

GandMarch, 2016-03-27:
@nathanpocock that's cool,thanks!!