OPCF-Members/UA-.NET-Legacy #18: GPL, RCL and .NET licence questions

ccasteyde, 2016-02-24:
The UA samples are dual licensed GPLv2 (not LGPL) and RCL.
I wonder how will be licensed any modification (GPL only or both?), because fixes made in the GPL context cannot be used in proprietary softwares that use the RCL license.
Moreover, as .NET assemblies are not really system libraries (since MS has open sourced them), I think the GPL license is incompatible with any .NET solution.

dk2129, 2016-02-24:
All fixes and accepted code submissions will only occur concurrent with the execution of the OPC [CLA](https://opcfoundation.org/license/cla/ContributorLicenseAgreementv1.0.pdf) (Contributor License Agreement) by the submitter.  This insures that all the code can be licensed under GPL2 and RCL.

GPL licensing of .NET code is allowed and commonly done.  For example, see the [mono-project](http://www.mono-project.com/docs/faq/licensing/).

closed 2016-02-24 by dk2129

ccasteyde, 2016-02-26:
This "http://opcfoundation.github.io/UA-.NET/" page indicates "Non-members must disclose their application code when using the UA .NET Stack."
Which implies members contributions can only be redistributed as RCL, not GPL. I think it should be clearly stated that CLA is required, otherwise there will be problems one day.
Moreover, you're right for the GPL licensing of .NET code. However, it has pitfalls.
It would be helpfull to state that redistribution of the Stack cannot be done together with .NET libraries (now easily feasible since .NET has been open sourced). This will not be a problem on Windows in general since .NET is provided by the system. But the licensing with GPL  of .NET code requires compatibility with all distributed libraries, and .NET libraries are not compatible if redistributed by the software. For instance: a commercial product that distributes and set up .NET Framework, together with the Stack under GPL used by a GPL tool (not the commercial product itself). The whole package would be illegal.