OPCF-Members/UA-.NET-Legacy #19: Unclear How To Get Started

narfanator, 2016-02-25:
It's not clear how to get started using this code. There's nothing in the ReadMe and the Getting Started guides on the project website require a paid membership.

Something as simple as "these files go here in this kind of project" would suffice.

dk2129, 2016-02-26:
Did you see the getting started paged on the landing page http://opcfoundation.github.io/UA-.NET/?

ccasteyde, 2016-02-26:
I think this page should also state that CLA is required even for Members otherwise members' contributions could not be distributed as GPL. See my comment on Issue 18.

jengstro2, 2017-03-07:
I'll throw my 5 cents here also (Yes, I'm early noob on OPC UA). This "Opc.Ua.CertificateGenerator.exe" cert creation tool is needed to launch/debug UA Configuration tool. Which "oddly" is the tool what you use to create certificates. I wonder why the sample code tries to generate certificate for the (cert) Configurator tool. I was having issue, that I got older version of this certgenerator.exe installed.  Which failed to extra cmd line parameter (-hashSize 0). Got sort it out by finding newer version of CertificateGenerator.exe (1.0.3). Which accepted the hashSize as 0.

Other thing is this %CommonDataFiles% env path value. What actually my understanding is returned from OS (from sample code) as C:\ProgramData via this sample code (where at least generated certs are created). Bit obscure for the beginners, due that this is "hidden" folder by default in Windows. It makes you wonder where the cert-folder physically is.

I think this sample UA .NET code project should contain everything what is needed in the solution itself. Having source+exe for CertificateGenerator.exe bundled into this sample.NET project (actually it shoud be just referenced as dll, now its executed as external proc/program from the code, which is not very elegant solution). Also, default xml-conf/cert folder could point relatively into project folder itself. Not to %CommonDataFiles%.

Sorry about the nag, but too noob on this to contribute yet. I'll explore more OPC UA first.

hrishiven, 2017-06-12:
Is there a tutorial on how to create the Boiler server and client?

tdesroches, 2017-09-26:
I also am looking for an ultra basic example server and client.  Perhaps a simple CLI program to demonstrate the basics.

astrowalker, 2017-09-27:
@tdesroches https://stackoverflow.com/questions/30573689/opc-ua-minimal-code-that-browses-the-root-node-of-a-server