OPCF-Members/UA-.NET-Legacy #22: Add Certificate Generator

dk2129, 2016-03-02:
The certificate generator is needed to run the samples. Therefore it should be added to this repository in binary format.

(current workaround - install the sample binaries first which implicitly also installs the certificate generator)

GandMarch, 2016-03-09:
you are right!
now i user an older  certificate generator,i got it from Siemens's Wincc 7.2.
but the old version of  certificate generator does not support "-hashsize" parameter,so you should wirte two slash before the the line "  writer.WriteLine("-hashSize {0}", hashSizeInBits);" in the file named "certificateFactory.cs",my english is very bad ,just see the picture below.i wish help you or somebody.
[image]

GandMarch, 2016-03-09:
where is sample binaries could download?
now i use an old version of certificate generator ,i got it from Siemens's WinCC 7.2

AlinMoldovean, 2016-03-09:
Hello,

dk2129, 2016-05-24:
The certificate generator is now available on github also: https://github.com/OPCFoundation/Misc-Tools. Information on how to use it for the samples is on the landing page (http://opcfoundation.github.io/UA-.NET/) - heading "Dependencies".

closed 2016-05-24 by dk2129