OPCF-Members/UA-.NET-Legacy #23: Opc.Ua.ConfiguredEndpoint Constructor does not respect provided configuration

MatKubicki, 2016-03-09:
There is a mistake in the file:
UA-.NET/Stack/Core/Stack/Configuration/ConfiguredEndpoints.cs

Which means that when you create a ConfiguredEndpoint using an ApplicationDescription and an EndpointConfiguration the EndpointConfiguration is always ignored.  A look through the samples suggests this is not intentional.  It is a simple fix by the looks of things.  Further down the constructor it checks for the member variable m_configuration being null, I suspect it should be checking the parameter configuration instead.

MatKubicki, 2016-03-09:
I've added a PR #24 for this.

MatKubicki, 2016-04-01:
Any response on this from anyone?  The change seems relatively simple and safe and it would be nice to see them pulled into the main codebase.  

A more general comment:
Some info on who is contributing to and managing this project, what testing/release milestones there are etc would be good.  I see Softing have committed a set of changes recently, have they been tested, should we treat the code in this repository as good to use or should we await an official release, if so how frequent will those be?

MatKubicki, 2016-05-20:
@AlinMoldovean I see you are at least able to push to this repository directly, and are the most recently active commiter/commenter any chance you could give this a once over, or at least let me know what the general plan is for public/OPC Foundation Members commits to this repository?

AlinMoldovean, 2016-05-23:
Fixed in commit d6a433f44fde57810524bba31eb6393c3382f012

closed 2016-05-23 by AlinMoldovean