OPCF-Members/UA-.NET-Legacy #26: Nuget

lemestrez, 2016-03-24:
Hello together,

what do you think about nuget? I think it could be useful to define nuget packages.
If you think it is usefull for this project, i would like to support you with this.

Best wishes

mpostol, 2016-03-24:
@lemestrez thanks for rising up this point. For me it is just must have option - no doubts. It is easy to create and greatly improves versioning. There is only one requirement that must be met, namely a firm versioning strategy - I recommend [Semantic Versioning 2.0.0](http://semver.org/). 

Good news is that as far as I know it is on the road map.

lemestrez, 2016-03-29:
I see in the AssemblyInfo. That the given version id looks meaningful like Semantic Versioning 2.0.0, but im not involed deeper in the OPC UA development.

I'm currently in a trial phase to learn OPC UA. So i want to build some sample applications for my solution.

AlinMoldovean, 2017-06-08:
NuGet packages will be added for the "UA-.NETStandardLibrary" project.

closed 2017-06-08 by AlinMoldovean