OPCF-Members/UA-.NET-Legacy #31: Scalability problem within request queue with multiple request threads

ccasteyde, 2016-04-07:
The request queue code awakes all the threads of its thread pool each time a new request is inserted in the queue. With the default configuration, under load due to several clients connecting or even while their publishing work, 100 threads are awaken for each incoming request. This can use all the CPU resources even on a Core i7.
The reason is the "m_event" is a ManualResetEvent instead of an AutoResetEvent. With an AutoResetEvent we could ensure all threads are effectively used simply by resetting the event if necessary.
Change proposal:
In RequestQueue constructor:

>                 // CODRA FFT 22624 - BEGIN 07/04/2016 - Issue GitHub#31
                m_event = new AutoResetEvent(false);
                // CODRA FFT 22624 - END

> 

In OnProcessRequestQueue:

>                             // CODRA FFT 22624 - BEGIN 07/04/2016 - Issue GitHub#31
                            // reset queue if queue is empty.
                            //if (m_queue.Count == 0)
                            //{
                            //    m_event.Reset();
                            //}
                            // restart a thread if queue is not empty.
                            if (m_queue.Count != 0)
                            {
                                m_event.Set();
                            }
                            // CODRA FFT 22624 - END

>

ccasteyde, 2016-04-08:
Better proposed fix in Issue #32.

AlinMoldovean, 2017-01-27:
The fix proposal was moved in Issue #32

closed 2017-01-27 by AlinMoldovean