OPCF-Members/UA-.NET-Legacy #32: Use after free in RequestQueue thread pool after Dispose

ccasteyde, 2016-04-07:
The RequestQueue Dispose code set the m_event and the Close it just afterwards, without waiting for all threads to complete.
The thread pool threads can therefore use the event after close, which can cause several problems (exceptions, bug check in underlying native code when accessing closed event, etc.).
This is not a good practice, either wait for all the threads to complete or transfer event.Close to the last thread:

> protected virtual void Dispose(bool disposing)
            {
                if (disposing)
                {
                    lock (m_lock)
                    {
                        m_stopped = true;
                        if (m_event != null)
                        {
                            m_event.Set();
                            m_event.Close();   // ARGGH
                            m_event = null;
                        }
                        m_queue.Clear();
                    }
                }
            }

Morever, with the proposed fix of issue 31, an alternative event for wakening all threads at dispose should be used.

ccasteyde, 2016-04-07:
Fix proposal: use a manualreset event for end request+OnProcessRequestQueue rewrite:
Members declaration:

>             // CODRA FFT 22624 - BEGIN 07/04/2016 - Issue GitHub#31 and #32
            private AutoResetEvent m_event;
            private ManualResetEvent m_endEvent;
            // CODRA FFT 22624 - END

ScheduleIncomingRequest (small logical fix):

>                         // CODRA FFT 22624 - BEGIN 07/04/2016 - Issue GitHub#31 and #32
                        if (m_totalThreadCount < m_minThreadCount || m_totalThreadCount == m_activeThreadCount)
                        // CODRA FFT 22624 - END

Dispose:

>            protected virtual void Dispose(bool disposing)
            {
                if (disposing)
                {
                    lock (m_lock)
                    {
                        if (!m_stopped)
                        {
                            m_stopped = true;

                            // CODRA FFT 22624 - BEGIN 07/04/2016 - Issue GitHub#31 and #32
                            //if (m_event != null)
                            //{
                            //    m_event.Set();
                            //    m_event.Close();
                            //    m_event = null;
                            //}
                            m_endEvent.Set();
                            // Close called at the last thread's termination.
                            m_endEvent = null;
                            m_event = null;
                            // CODRA FFT 22624 - END

                            m_queue.Clear();
                        }
                    }
                }
            }

OnProcessRequestQueue:
            private void OnProcessRequestQueue(object state)
            {
                // CODRA FFT 22624 - BEGIN 07/04/2016 - Issue GitHub#31 and #32
                ManualResetEvent hEnd = null;
                AutoResetEvent hEvt = null;
                lock (m_lock)
                {
                    hEnd = m_endEvent;
                    hEvt = m_event;
                }
                if (hEnd != null)
                {
                    WaitHandle[] events = new WaitHandle[] { hEnd, hEvt };

                    bool bEnd = false;
                    do
                    {
                            int iEvent = WaitHandle.WaitAny(events, 30000);
                            switch (iEvent)
                            {
                                case WaitHandle.WaitTimeout:
                                    lock (m_lock)
                                    {
                                        if (m_totalThreadCount > m_minThreadCount)
                                        {
                                            m_totalThreadCount--;
                                            bEnd = true;
                                        }
                                    }
                                    break;
                                case 0: // End request:
                                    lock (m_lock)
                                    {
                                        m_totalThreadCount--;
                                        if (0 == m_totalThreadCount)
                                        {
                                            hEnd.Close();
                                            hEvt.Close();
                                        }
                                        bEnd = true;
                                    }
                                    break;
                                case 1: // New request:
                                    IEndpointIncomingRequest request = null;

                                    lock (m_lock)
                                    {
                                        // check if the queue is empty.
                                        if (m_queue.Count == 0)
                                        {
                                            continue;
                                        }

                                        m_activeThreadCount++;

                                        request = m_queue.Dequeue();

                                        // restart a thread if queue is not empty.
                                        if (m_queue.Count != 0)
                                        {
                                            hEvt.Set();
                                        }
                                    }

                                    // process the request.
                                    try
                                    {
                                        m_server.ProcessRequest(request, state);
                                    }
                                    catch (Exception e)
                                    {
                                        Utils.Trace(e, "Unexpected error processing incoming request.");
                                    }

                                    lock (m_lock)
                                    {
                                        m_activeThreadCount--;
                                    }
                                    break;
                            }
                    } while (!bEnd);
                }
                // CODRA FFT 22624 - END
            }

cmorty, 2016-06-30:
@ccasteyde: My experience with other projects is that there is often more awareness if you make a pull-request.

AlinMoldovean, 2017-01-27:
The reported issue is from an older version of the RequestQueue.Dispose() implementation.
The actual code uses Monitor.PulseAll(m_lock) instead of m_event.Set()/m_event.Close().

	protected virtual void Dispose(bool disposing)
	{
		if (disposing)
		{
			lock (m_lock)
			{
				m_stopped = true;

				Monitor.PulseAll(m_lock);

				m_queue.Clear();
			}
		}
	}

closed 2017-01-27 by AlinMoldovean