OPCF-Members/UA-.NET-Legacy #33: Wrong revised MawKeepAliveCount and LifeTimeCount calculation

ccasteyde, 2016-04-07:
The standard server sample does wrong calculations when the publishing interval is revised, because it still computes the time limits with the revised publishing interval instead of the requested one. This can lead in particular to actual subscription lifetime greater than the max subscription lifetime configured. The bug is both in SubscriptionManager.cs, CreateSubscription and ModifySubscription.

> revisedPublishingInterval = CalculatePublishingInterval(requestedPublishingInterval);
            // calculate the keep alive count.
            revisedMaxKeepAliveCount = CalculateKeepAliveCount(revisedPublishingInterval, requestedMaxKeepAliveCount);
            // calculate the lifetime count.
            revisedLifetimeCount = CalculateLifetimeCount(revisedPublishingInterval, revisedMaxKeepAliveCount, requestedLifetimeCount);

for example: revised publishing interval is 1000 instead of 100. CalculateKeepAliveCount receives revised publishing interval and keepAliveInterval is 10x too large. It may be capped.
Same for LifeTimeCount. Moreover, LifeTimeCount can be increased to 3*KeepAliveCount after check against MaxSubscriptionLifeTime, even if it should not have been caped in the first place and was correct from the client point of view.

Proposal: Calculate requested time interval and revise KeepAliveCount and LifeTimeCount if publishing interval is revised, before applying configuration checks and max.

ccasteyde, 2016-04-07:
Fix proposal:
SubscriptionManager.cs, SubscriptionManager constructor:

>             // CODRA FFT 22623 - BEGIN 07/04/2016, GitHub 33
            m_minPublishingInterval      = (uint)configuration.ServerConfiguration.MinPublishingInterval;
            m_maxPublishingInterval      = m_minPublishingInterval;
            if ((uint)configuration.ServerConfiguration.MaxPublishingInterval > (uint)configuration.ServerConfiguration.MinPublishingInterval)
                m_maxPublishingInterval  = (uint)configuration.ServerConfiguration.MaxPublishingInterval;
            m_publishingResolution       = configuration.ServerConfiguration.PublishingResolution;
            m_minSubscriptionLifetime    = (uint)configuration.ServerConfiguration.MinSubscriptionLifetime;
            m_maxSubscriptionLifetime    = m_minSubscriptionLifetime;
            if ((uint)configuration.ServerConfiguration.MaxSubscriptionLifetime > (uint)configuration.ServerConfiguration.MinSubscriptionLifetime)
                m_maxSubscriptionLifetime    = (uint)configuration.ServerConfiguration.MaxSubscriptionLifetime;
            // CODRA FFT 22623 - END

CreateSubscription and ModifySubscription:

>             // CODRA FFT 22623 - BEGIN 07/04/2016, GitHub 33
            // calculate the keep alive count.
            revisedMaxKeepAliveCount = CalculateKeepAliveCount(revisedPublishingInterval, requestedPublishingInterval, requestedMaxKeepAliveCount);

            // calculate the lifetime count.
            revisedLifetimeCount = CalculateLifetimeCount(revisedPublishingInterval, revisedMaxKeepAliveCount, requestedPublishingInterval, requestedLifetimeCount);
            // CODRA FFT 22623 - END

CalculateKeepAliveCount:

>         // CODRA FFT 22623 - BEGIN 07/04/2016, GitHub 33
        protected virtual uint CalculateKeepAliveCount(double publishingInterval, double requestedPublishingInterval, uint keepAliveCount)
        {
            // set default.
            if (keepAliveCount == 0)
            {
                keepAliveCount = 3;
            }

            double keepAliveInterval = keepAliveCount * requestedPublishingInterval;
            keepAliveCount = (uint)(keepAliveInterval / publishingInterval + 0.5);
            // CODRA FFT 22623 - END

CalculateLifeTimeCount:

>         // CODRA FFT 22623 - BEGIN 07/04/2016, GitHub 33
        protected virtual uint CalculateLifetimeCount(double publishingInterval, uint keepAliveCount, double requestedPublishingInterval, uint lifetimeCount)
        {
            double lifetimeInterval = lifetimeCount * requestedPublishingInterval;
            lifetimeCount = (uint)(lifetimeInterval / publishingInterval + 0.5);
            // CODRA FFT 22623 - END

cristipogacean, 2016-09-13:
The revised are application speciffic and any application can decide to use the onw values. The server commits to the revised values, and it is application responsibility to make sure the various variants are consistent. we cannot enforce restrictions on the stack side.

closed 2017-06-08 by AlinMoldovean