OPCF-Members/UA-.NET-Legacy #34: .NET Sample UA client ExpandedNodeId Issue for nodes that have namespaceuri

Vaageesh, 2016-04-14:
Using the Java stack to develop the OPC UA Server. But .NET Sample client is not able convert ExpandedNodeId to nodeId for those node which have the namespaceuri "http://opcfoundation.org/UA/".

Error log :
6452 - 10:00:30.941 Could not fetch references for valid node with NodeId = i=85. Error = Cannot cast an absolute ExpandedNodeId to a NodeId. Use ExpandedNodeId.ToNodeId instead.
6452 - 10:00:30.955 Could not fetch references for valid node with NodeId = i=35. Error = Cannot cast an absolute ExpandedNodeId to a NodeId. Use ExpandedNodeId.ToNodeId instead.
6452 - 10:00:30.960 Could not fetch references for valid node with NodeId = nsu=http://opcfoundation.org/UA/;i=2004. Error = Cannot cast an absolute ExpandedNodeId to a NodeId. Use ExpandedNodeId.ToNodeId instead.
6452 - 10:00:30.967 Could not fetch references for valid node with NodeId = nsu=http://opcfoundation.org/UA/;i=61. Error = Cannot cast an absolute ExpandedNodeId to a NodeId. Use ExpandedNodeId.ToNodeId instead.

K47S, 2016-05-09:
Hello Vaageesh,
what exactly are you trying to do?

If you need for example the NodeId for the ObejctsFolder you can use: 
Softing.Opc.Ua.Sdk.ObjectIds.ObjectsFolder

If you have a ExpandedNodeId (yourExpandedNodeId) and you want the corresponding NodeId you can use:
var nodeId = ExpandedNodeId.ToNodeId(yourExpandedNodeId , null);
If this doesn't work you can try:
var nodeId = new NodeId(yourExpandedNodeId.Identifier, yourExpandedNodeId.NamespaceIndex)

AlinMoldovean, 2017-06-08:
As @K47S mentioned, the ExpandedNodeId.ToNodeId() method should be used for casting the absolute ExpandedNodeId to a NodeId

You can also use the set of predefined constant NodeIds.

closed 2017-06-08 by AlinMoldovean