OPCF-Members/UA-.NET-Legacy #35: Client/Stack appends a forward slash to endpoint URLs

kevinherron, 2016-04-21:
This was discovered here in Phoenix at the OPC interop.

It appears the sample client, perhaps due to some fault in the stack, since it's affecting users of the .NET stack as well, appends a forward slash to the end of endpoint URLs before it connects to a server.

For example, instructing the client to connect to `opc.tcp://192.168.10.17:12685` will result in a Hello message containing the endpoint URL `opc.tcp://192.168.10.17:12685/`.

BoeseB, 2016-05-10:
Does the server not respond due to the trailing slash or is it just you dislike the trailing slash?
A slash indicates a directory to the server, since a single hostname is in my opinion similar to a directory and not like a file, the trailing slash should be ok.

An example why the trailing slash matters in Webservers:
In SEO there is a huge ongoing debate if you should add or remove the trailing slash from URLs
if you address a directory you should add a trailing slash e.g. http://myawesomesite.com/services/ 
and the webserver will look for the default file in this directory. 
if you call http://myawesomesite.com/services without the trailing slash the webserver first should search for a file called "services" then if it doesn't finde the file it may throw an error or if it is fault tolerant search for the "services" directory and open the default file in it.

Either way as far as i remember the source of the trailing slash is the Uri-Class of the .NET Framework.

AlinMoldovean, 2016-05-10:
The System.Uri class of the .NET Framework generates the trailing slash.

	Uri uri = new Uri(url);
	
	EndpointDescription description = new EndpointDescription();
	description.EndpointUrl            = uri.ToString();

I don't know what is wrong with that, but it should not generate any issue on the server side.
The path component of the URI must begin with a single slash (/) if an authority part was present, but must not begin with a double slash.
	scheme:[//[user:password@]host[:port]][/]path[?query][#fragment]

kevinherron, 2016-05-10:
This server is strict about the endpoint URL it receives and does treat receiving `opc.tcp://192.168.10.17:12685/` the same as receiving `opc.tcp://192.168.10.17:12685`.

BoeseB, 2016-05-11:
So after some further research, i come to conclusion this should be a bug of the server.
The Reason is the two URIs you provided should be treated the same according to RFC 3986  and RFC 2616. The first one is about URI Syntax in general and the second one is about HTTP.

[RFC 3986](http://www.ietf.org/rfc/rfc3986.txt)
> 6.2.3.  Scheme-Based Normalization
>    The syntax and semantics of URIs vary from scheme to scheme, as
>    described by the defining specification for each scheme.
>    Implementations may use scheme-specific rules, at further processing
>    cost, to reduce the probability of false negatives.  For example,
>    because the "http" scheme makes use of an authority component, has a
>    default port of "80", and defines an empty path to be equivalent to
>    "/", the following four URIs are equivalent:
> 
>       http://example.com
>       http://example.com/
>       http://example.com:/
>       http://example.com:80/

or 

[RFC 2616](http://www.rfc-editor.org/rfc/rfc2616.txt).
> 3.2.3 URI Comparison
>    When comparing two URIs to decide if they match or not, a client
>    SHOULD use a case-sensitive octet-by-octet comparison of the entire
>    URIs, with these exceptions:
> 
>       - A port that is empty or not given is equivalent to the default
>         port for that URI-reference;
> 
>         - Comparisons of host names MUST be case-insensitive;
> 
>         - Comparisons of scheme names MUST be case-insensitive;
> 
>         - An empty abs_path is equivalent to an abs_path of "/".
> 
>    Characters other than those in the "reserved" and "unsafe" sets (see
>    RFC 2396 [42]) are equivalent to their ""%" HEX HEX" encoding.
> 
>    For example, the following three URIs are equivalent:
> 
>       http://abc.com:80/~smith/home.html
>       http://ABC.com/%7Esmith/home.html
>       http://ABC.com:/%7esmith/home.html

kevinherron, 2016-05-11:
Both of your citations are specific to HTTP.

kevinherron, 2016-05-11:
FWIW, I _am_ going to add logic to handle this case in the Java stack, but I still think you should do something about this because 

1. .NET-based clients are the only clients that have demonstrated this issue
2. it's a poor experience for the user that they have no control over the endpoint URL that gets sent, although it may just be a UI problem

BoeseB, 2016-05-11:
I agree with you that it's somehow strange that the URI is changed from `http://example.com` to
`http://example.com/`. I had the same problem in the UI of our applications browser for OPC-UA endpoints. I was wondering where the trailing slash is coming from, but in my case it was just a cosmetic problem so far, cause i never experienced any problems with the opc servers we used for testing the application. They all treated the URI with trailing slash the same as without.

The folks of this C# stack can do little about the URI class appending this trailing slash that is appended to any absolute path, which isn't defining any sub directories. The only options would be to not use the .NET Frameworks URI class and define their own class for validating the uri or ignore validation and go with a simple string.

Sorry if the citations all are taking the http scheme as examples, i am by no means an expert on URI Schemes and maybe the opc.tcp sheme could specify this case in another way. 

I am no member of the OPC Foundation so please don't take my opinions for their word ;) 
I was just intrested in this Issue cause of my own experience with the trailing slash and if i should consider it a bug in the .NET Frameworks URI implementation.

YakupIpek, 2016-06-08:
I also hit the bug in my application just because i assumed that url == new Uri(url).ToString() is true.

closed 2016-09-13 by cristipogacean