OPCF-Members/UA-.NET-Legacy #36: An examle of how to use OPC services

MikhailKrasnobaev, 2016-04-28:
Can someone provide a simple example of how to use UA.NET services from the sample apps?
What I need is to read data from a tag on an Kepware OPC server. Thanks in advance.

K47S, 2016-04-28:
Have you already looked in the documentation?
The Kepware OPC server is a OPC Classic Server isn't it?

Documentation:
http://opcfoundation.github.io/UA-.NET/help/index.htm#uaconfig_manage_com_interop.htm#Wrap_COM_Servers

MikhailKrasnobaev, 2016-04-28:
> Have you already looked in the documentation?

Yes, sure, I even tried going through the OPC UA specificaton documentation.

> The Kepware OPC server is a OPC Classic Server isn't it?

No, I'm trying to use OPC UA.

> Documentation:

Basically I need this functionality http://opcfoundation.github.io/UA-.NET/help/reading_nodes.htm 
But without the GUI.

awcullen, 2016-05-07:
This post might be helpful.
http://stackoverflow.com/questions/30573689/opc-ua-minimal-code-that-browses-the-root-node-of-a-server

rzvpopescu, 2016-05-30:
Hello ,
Do you have any examples of how to write a value for a writable item from and OPC Server , without the GUI ?

K47S, 2016-05-30:
1. Create a Session
2. Find out NodeId (intWriteNodeId)
3. Use session.Write
`
WriteValueCollection writeValues = new WriteValueCollection();
// Int32 Item
WriteValue intWriteVal = new WriteValue();
intWriteVal.NodeId = new NodeId(intWriteNodeId);
intWriteVal.AttributeId = Attributes.Value;
intWriteVal.Value = new DataValue(intValue);

writeValues.Add(intWriteVal);
// Call the Write service
ResponseHeader responseHeader = session.Write(null,
                                                            writeValues,
                                                            out results,
                                                            out diagnosticInfos);

`

rzvpopescu, 2016-05-30:
Thanks K476 !

MikhailKrasnobaev, 2016-05-30:
Hi, additionally to the comment from K47S you can play around with this project https://github.com/MikhailKrasnobaev/OPC-UA.NET-ConsoleApp . I stripped down the source code from OPC UA to two libraries. To read out or write a value to a tag it works, for a subscription - do not create a session using  "using()" :), create a global variable.

ps. thanks to @awcullen  for the link to stackoverflow

K47S, 2016-05-30:
@MikhailKrasnobaev 
nice to see that you opened a repo for your project and share your code with the community.
Is it possible for you to mark this issue as resolved now?

rzvpopescu, 2016-05-30:
@MikhailKrasnobaev  Thanks for your help. I managed to read tags, and their values, using an approach like yours.
Another question : It is possible to start a session without having a certificate ? I couldn't find a way to bypass certificate validation.

mpostol, 2016-05-30:
Hi guys,
Can we move discussion like this to [Home of developer communities]( https://gitter.im).
The issues here should be related to code development, otherwise it will be overloaded soon.

K47S, 2016-05-30:
@mpostol 
Is there an existing channel on gitter for this repo? Couldn't find one.

just one last answer:
@rzvpopescu 

ApplicationConfiguration of your Client:

ApplicationConfiguration .CertificateValidator.CertificateValidation += CertificateValidator_CertificateValidation;

private void CertificateValidator_CertificateValidation(CertificateValidator sender, CertificateValidationEventArgs e)
{
Console.ForegroundColor = ConsoleColor.Red;
Console.WriteLine("Security issue! Certificate accepted without any checks!");
Console.ResetColor();
e.Accept = true;
}

MikhailKrasnobaev, 2016-05-30:
@rzvpopescu well, no. According to the documentation there is a security certificate setting NONE (in the OPC UA source it is a default for the client), it means you just need to have a valid certificate for the client and on the server side security setting should allow NONE. My example auto-excepts  the server certificate.

closed 2016-05-30 by MikhailKrasnobaev

rzvpopescu, 2016-05-30:
Thanks everyone for your help. As @mpostol  suggested, it would be nice to have a gitter channel.