OPCF-Members/UA-.NET-Legacy #51: Certificate Issue BadCertificateHostNameInvalid

MatKubicki, 2016-07-29:
Hi,

We are using the Opc.Ua.Client library to connect to various OPC UA Servers (Ignition and KepServer to name two).

To take KepServer as an example we find that it creates a server certificate containing its fully qualified domain name and therefore if we attempt to connect via IP address we get a BadCertificateHostInvalid failure.  This still happens if we setup KepServer with only IP address endpoints and regenerate the certificate.

Looking at the code for the Opc.Ua.Client I see that there is an option on the method Opc.Ua.Client.Session.Create called 'checkDomain', this disables the check for the domain in the certificate within the Create method, but is not passed on to Session.Open where another ceck is performed (in another call to 'CheckCertificateDomain').

Is this a bug?  Looking at other OPC UA clients I see at least some (Unified Automation) have an option to disable the domain check, should we allow connections in this situation and change the Opc.Ua.Client library to allow it?

Thanks,

Matt

MatKubicki, 2016-07-29:
From looking at the spec and some more digging around the sample code I'm convinced this is a bug in the Session.Create method and an omission from the Session.Open method.

Hopefully others agree and we can pull this in.

AlinMoldovean, 2016-08-03:
Fixed with Pull #52

closed 2016-08-03 by AlinMoldovean