OPCF-Members/UA-.NET-Legacy #65: Add support for GDS

cristipogacean, 2016-09-13:
RegisterServer2 respective FindServersOnNetwork support to be added to stack and exemplified in the sample applications.

AlinMoldovean, 2016-10-28:
Related classes and support for FindServersOnNetwork and RegisterServer2 is present not in the stack.
The implementation for automatic calls for these services should be postponed until there is an official release of LDS-ME(responding the service calls) .

AlinMoldovean, 2017-01-26:
Support for FindServersOnNetwork and RegisterServer2 was added in sample applications in commit 4e6b3f3546a5407adff00895c7da01d7b430424f and commit 9ad6eb7c4a74c747a1a40c6267177ea27fc95dff
The server attempts to register using a "RegisterServer2" call and falls back to "RegisterServer" in case it doesn't succeed.
UA Sample Client provides now support for discovery using FindServersOnNetwork.
Sample Server and Sample Client was used for testing the beta version of LDS-ME

closed 2017-01-26 by AlinMoldovean