OPCF-Members/UA-.NET-Legacy #66: An exception occurs in NodeCache.FetchNode.

iPuerto, 2016-09-28:
I found that An exception occurs in NodeCache.cs line 645.
When ExpandedNodeId.IsAbsolute is true and converts an ExpandedNodeId to a NodeId, an exception occurs.

I think that should set the node Id obtained by converting the namespace URI to the server dependent index in NodeCache.cs line 645.

For example, Add a constructor to the Node class;

```
public partial class Node : IFormattable, ILocalNode
{
    #region Constructors
------
    /// <summary>
    /// Creates a node from the local node id and a reference desciption.
    /// </summary>
    /// <param name="localId">The local node id.</param>
    /// <param name="reference">The reference.</param>
    public Node(NodeId localId, ReferenceDescription reference)
    {
        Initialize();

        m_nodeId = localId;
        m_nodeClass = reference.NodeClass;
        m_browseName = reference.BrowseName;
        m_displayName = reference.DisplayName;
    }
-----
```

Change the NodeCache.cs line 645;

`Node target = new Node(reference);`
↓ 
`Node target = new Node(localId, reference);`

Thanks.

dk2129, 2016-09-28:
Can you please make a pull request with these changes? Once you sign the contributor license we can merge your proposed changes.

iPuerto, 2016-09-28:
OK. I will make a pull request.

commit 7e4c1973 (2017-05-03): Fixes #66 - Reference containing an absolute ExpandedNodeId throws exception in Node(ReferenceDescription reference) constructor.
closed 2017-05-03 by AlinMoldovean