OPCF-Members/UA-.NET-Legacy #69: Consider rename of 'Opc.Ua.NodeState.Description' to prevent member hiding?

Miragecoder, 2016-09-30:
In reference to:
https://github.com/OPCFoundation/UA-.NET/blob/master/Stack/Core/Stack/State/NodeState.cs#L334

I believe that **'Description'** is a very common property name to have on an object type. Using that name on an OPC object type will however result in member hiding. 

For example, if the following **'ObjectType'**-declaration is fed through the modelcompiler...

```
<opc:ObjectType SymbolicName="MyObject" >
	<opc:Children>
		<opc:Property SymbolicName="Name" ModellingRule="Mandatory" DataType="ua:String" AccessLevel="ReadWrite"  />
		<opc:Property SymbolicName="Description" ModellingRule="Mandatory" DataType="ua:Int32" AccessLevel="ReadWrite"  />
	</opc:Children>
	<opc:References />
	<opc:ClassName />
	<opc:Fields />
</opc:ObjectType>
```

...it will generate a **'PropertyState<string> Description'**-property on the **'ObjectTypeState'** generated for this declaration, which hides **'Opc.Ua.NodeState.Description'** and causes a compiler warning. Wouldn't it make more sense to change **'Opc.Ua.NodeState.Description'** into **'Opc.Ua.NodeState.LocalizedNodeDescription'**? (Or something similiar?) 

Obviously this would lead to breaking changes but at least I want to put my thoughts regarding this design out here.

AlinMoldovean, 2016-10-11:
"Description" is one of the standard OPC UA node attributes just like NodeId, NodeClass, BrowseName, DisplayName...\
It is pretty much late to take in consideration changing it's name.

AlinMoldovean, 2017-05-03:
Breaking changes cannot be added at this point of development.

closed 2017-05-03 by AlinMoldovean