OPCF-Members/UA-.NET-Legacy #72: TransferSubscriptions currently not implemented?

Miragecoder, 2016-10-05:
Judging from the current state of the 'TransferSubscriptions'-method:

https://github.com/OPCFoundation/UA-.NET/blob/fedc75497a3d83529da3a3ca65ce0bd1661ff4bb/SampleApplications/SampleLibraries/Server/Subscription/SubscriptionManager.cs#L1074

...I'm under the impression that it's currently not supported. We're currently evaluating usage of this SDK but we require the ability to recover notifications from a subscription that belong to a dead session, for which I believe we rely on the ability to transfer subscriptions to different sessions. Is it correct that this feature is not available at this time?

AlinMoldovean, 2016-10-05:
Yes, you are right. For the moment TransferSubscriptions is not supported and it requires implementation.

Miragecoder, 2016-10-06:
That's unfortunate but understandable. Is there any chance that we might see such an implementation sometime in the near future? I believe that there's not such a thing as a roadmap for this project, is there? From what I have been able to collect I believe this is also one of the more trickier-to-implement functionality of the standard.

cristipogacean, 2016-10-06:
There is a roadmap, but TransferSubscriptions is not planned for the near future. You are wellcome to implement the functionality and contribute it to the project. Alternatively you can consider a commercial toolkit.

AlinMoldovean, 2017-05-03:
Support for TransferSubscriptions  is not scheduled yet.

closed 2017-05-03 by AlinMoldovean