OPCF-Members/UA-.NET-Legacy #74: NotImplementedException on InvokeService method called connecting from official OPC-UA Sample Client

Rowandish, 2016-10-07:
Hi all,
**Target:**
I'd like to develop a OPC-UA server; for the beginnging I'd like to use your sample WCF server 
(_SampleApplications\WCF\Server_) in order to connect to the official OPC-UA sample client.

**What I did**:
I installed the server certificate on my pc and the server works and answer correctly to your sample client (_SampleApplications\WCF\Client_).

**Problem**:
When I try to connect to it from the official OPC-UA sample client I have `NotImplementedException` on `InvokeService` method.
Here are two screenshot:

[image]

[image]

What should I do or implement in order to make the server answer correctly to the client?
Many thanks!

AlinMoldovean, 2016-10-11:
Hello.
You can use the ReferenceServer sample project as a starting point.
The WCF Client/Server samples have some custom implementation and they do not use the standard UA Stack.

Rowandish, 2016-10-12:
Ok, I'll have a look. Thanks

closed 2016-10-12 by Rowandish