OPCF-Members/UA-.NET-Legacy #76: How to configure the certificate date rules in Data Access server?

panhenry, 2016-10-11:
The Quickstart Data Access Server in Dashboard failed the 007.js and 008.js in Security Certificate Validation CTT items.
If we want to configure the certificate date rules in server that we can launch from Dashboard, Could you help to explain how to configure the certificate date rules in Quickstart Data Access Server?

Thank you.

[image]

AlinMoldovean, 2016-10-11:
These specific tests require the "expired.der" and "notyetvalid.der" certificates from the CTT project to be copied in the trust list of the server before running them.
Maybe his is the reason why the tests are failing.

panhenry, 2016-10-17:
Thank you for your reply.
Do you mean that we put the certification files in the wrong path?

If we do not check the DisableCertificateTimeInvalid option, we can pass the 007.js and 008.js items.
Can you pass the 007.js and 008.js with checking the DisableCertificateTimeInvalid option?

AlinMoldovean, 2016-10-18:
Hello,
Yes, those tests should pass the compliance testing without checking the DisableCertificateTimeInvalid option.
But first, before running the CTT testing against an OPC UA Server there is an initial step of copying some certificates from the folder of the CTT project into the trust list of the server.
In case of Quickstart Data Access Server or better Quickstart Reference Server (because that is the official server used for compliance testing) you need to copy the following certificates from CTTProjectFolder\PKI\CA\certs into %CommonApplicationData%\OPC Foundation\CertificateStores\UA Applications:
expired.der
NotyetValid.der
opcuactt.der
opcuactt_ca.der
opcuactt_incorrectAppUri.der
opcuactt_incorrectIP.der
opcuactt_incorrectSign.der
opcuactt_issued.der
opcuactt_revoked.der
opcuser.der

closed 2016-10-18 by AlinMoldovean