OPCF-Members/UA-.NET-Legacy #77: OPC UA dll for .NET framework version 4.0

AntoniPancras, 2016-10-18:
Hi,
This is not an issue. Asking a solution for my problem given below.

One of Our OPC Client product is already using Opc.Ua.Client.dll and Opc.Ua.Core.DLL . But it is very old version 1.01.324.
 We want to upgrade to the latest version 1.0.3.340. Looks the latest version of OPC UA dlls are built with target framework version 4.5.
 Basically i need .Net framework version 4.0 of OPC UA dlls. because my OPC client application has the setting with target framework 4.0 in visual studio 2015.

I am planning  to do something  like given below.
 Please correct me if i am wrong.

1. Dowload the source code from github
 2. Open Visual studio 2015, set the target framwork to 4.0
 3. Build the dlls
 4. Refer those dlls in my application

Thanks

Antoni

cristipogacean, 2016-10-18:
Basically the steps are correct. The Interface has not chaned significantly so probably you only need to fix a few things in your application.. 
Please doublecheck the licensing coditions for the stack because that was changed in the meanwhile for non OPC foundation members.

AlinMoldovean, 2016-10-18:
Hello Antoni,
The initial version you were using was build with target .NET Framework 3.5.
Afterwards the Visual Studio projects were updated to .NET 4.5.1 but basically the code should compile with any Framework starting from 3.5

AntoniPancras, 2016-10-20:
Thanks for the quick reply. I appreciate it

Thanks
Antoni

closed 2016-10-20 by AlinMoldovean