OPCF-Members/UA-.NET-Legacy #79: What is the purpose of setting server timestamp on write by client?

astrowalker, 2016-11-10:
According to spec "The serverTimestamp is used to reflect the time that the Server received a Variable value or knew it to be accurate" (part 4, section 7.7.4).

But when writing value by a client you can set both source timestamp (OK) and server timestamp. I don't understand for what purpose client sets server timestamps -- after all, client!=server, and it can be only a time of sending data, not receiving.

AlinMoldovean, 2016-11-10:
ServerTimestamp is simply transported together with DataValue structures, but it shouldn't be set by clients when performing Write service calls.

Also, writing ServerTimestamp is not supported in the default implementation of the UA .NET Server stack.
The servers will reject all Write requests where ServerTimestamp is specified (i.e. different than DateTime.MinValue)

	// writes to server timestamp never supported.
	if (value.ServerTimestamp != DateTime.MinValue)
	{
		return StatusCodes.BadWriteNotSupported;
	}

astrowalker, 2016-11-10:
Thank you very much for explanation.

closed 2016-11-10 by astrowalker