OPCF-Members/UA-.NET-Legacy #81: Is it possible to write async method in server?

astrowalker, 2016-11-22:
Hello everyone :-),

  I can see I can call a method from a client asynchronously, but I don't see how a method itself can be handled asynchronously inside the server. The OnCall events are synchronous ones -- so it is possible (and yes, how) to write asynchronous method on server side?

AlinMoldovean, 2016-11-23:
On the server side the requests are processed synchronously even if they are called asynchronously by clients.
The exception is the "Publish" request. This is processed asynchronously.

        protected override void ProcessRequest(IEndpointIncomingRequest request, object calldata)
        {
            if (request.Request.TypeId != DataTypeIds.PublishRequest)
            {
                request.CallSynchronously();
                return;
            }

            // set the context.
            SecureChannelContext.Current = request.SecureChannelContext;

            try
            {
                // begin a new publish.
                if (request.Calldata == null)
                {
                    BeginPublish(request);
                }

                // continue a previously queued publish.
                else
                {
                    CompletePublish(request);
                }
            }
            catch (Exception e)
            {
                request.OperationCompleted(null, new ServiceResult(e));
            }
        }

So you can see an example of asynchronous implementation in the Publish mechanism.
Starting point is StandardServer.BeginPublish(IEndpointIncomingRequest request)

astrowalker, 2016-11-23:
@AlinMoldovean , thank you very much for the confirmation.

closed 2016-11-23 by astrowalker