OPCF-Members/UA-.NET-Legacy #83: FormatException occurs when loading the Aggregate Tester data

esugoi, 2016-11-29:
When loading the HA Aggregate Tester application data for the first time, there is an exception being throw on the ValidateTimestamp method:

```
"The second must be less than 60."
at Quickstarts.TestData.ValidateTimestamp(Object formattedValue) in F:\github\opcua\UA-.NET\SampleApplications\Workshop\HistoricalAccess\Tester\TestDataHelpers.cs:line 407
at Quickstarts.TestData.ToDataValues(ValueType[] values) in F:\github\opcua\UA-.NET\SampleApplications\Workshop\HistoricalAccess\Tester\TestDataHelpers.cs:line 527
at Quickstarts.TestData.GetProcessedValues(ProcessedDataSetType dataset) in F:\github\opcua\UA-.NET\SampleApplications\Workshop\HistoricalAccess\Tester\TestDataHelpers.cs:line 256
at Quickstarts.MainForm.TestNameCB_SelectedIndexChanged(Object sender, EventArgs e) in F:\github\opcua\UA-.NET\SampleApplications\Workshop\HistoricalAccess\Tester\MainForm.cs:line 1201
```

This is happening when parsing the seconds from the timestamp of the Processed data values. The timestamp is in the format bellow (with milliseconds set):
_00:00:05.000_

As opposed to the Raw data values (without milliseconds set):
_00:00:05_

The problem is that, using my current locale, the converting of the seconds to double is transforming all of the 5.000 to the value 5000, an thus resulting on the behavior above.

SparhawkSoftware, 2016-11-29:
Localization was not a priority when developing this tool so such issues are not a surprise.

Note that the version which is used to manage the specification is found here: https://github.com/OPCFoundation/Misc-Tools

If you have fixes that can resolve the localization then please submit them on that repo.

esugoi, 2016-11-29:
I see...
I think I pulled the request on the wrong repo, sorry about that.
Thanks for the answer, Randy!

commit 144a2c9f (2016-12-05): Fix #83. Parsing the milliseconds from timestamp of testing data.
commit bcfe5841 (2016-12-05): Merge pull request #84 from esugoi/master

Fix #83. Parsing the milliseconds from timestamp of testing data.
closed 2016-12-05 by AlinMoldovean