OPCF-Members/UA-.NET-Legacy #85: How to define and use custom structure type?

astrowalker, 2016-12-02:
Say I would like to define type `Person` with "fields" `Name` and `Age` in terms of OPC-UA. Also I would like to define variable node `person` of type `Person` and write to that node in one step. **How to do it?**

Please note: when I update `person` with such data `(Kevin, 47)`, `(Jane, 22)` client should get in subscription, or when reading the variable directly only those 2 pairs, not something like `(Kevin,22)`.

SparhawkSoftware, 2016-12-02:
From a specification perspective you would:

Create a DataType (PersonDataType). It is a subtype of Structure.
Use that DataType as the DataType for a Variable.
When reading the Value attribute you would get an ExtensionObject (a.k.a. Structure).
The ExtensionObject would contain a serialized version of PersonDataType.
For writing you would create an ExtensionObject with a PersonDataType and write to the Value attribute.

The UA Sample Library project has numerous examples of the above in TestData folder.

astrowalker, 2016-12-05:
@SparhawkSoftware , thank you!

For clarification, "Create a DataType (PersonDataType)" -- as a node in ObjectTypes, correct?

"The UA Sample Library project has numerous examples of the above in TestData folder." Do you know any straightforward example? Because all I found is ["TestData.DataTypes" file](https://github.com/OPCFoundation/UA-.NET/blob/master/SampleApplications/Samples/Common/TestData/TestData.DataTypes.cs) which creates randomly single field structure (at least this is my understanding) -- still reading this one example. I didn't find any other.

benjamin1988, 2017-12-18:
Is their any possibility to rebuild the custom server data types at client side? 
Now i have to rebuild all custom types in the client application, it would be nice to have a generic function to create a data type with the definition of an encodable ExtensionObject. Or even with a TypeDefinition of a Object Type.

ZbynekZ, 2018-02-13:
For clients written in .NET, QuickOPC has support for custom types (OPC UA "Complex data"). It can decode and encode the extension objects for you, automatically. See [QuickOPC 2018.1 release announcement](http://www.opclabs.com/resources/news-archive/1154-quickopc-2018-1-released) and [OPC UA Complex Data Extensions documentation](http://opclabs.doc-that.com/files/onlinedocs/QuickOpc/2018.1/User%27s%20Guide%20and%20Reference-QuickOPC/webframe.html#OPC%20UA%20Complex%20Data%20Extension.html).