OPCF-Members/UA-.NET-Legacy #89: Session lifetime management pattern (UA Core Library)?

maiorfi, 2016-12-08:
Hi.

What is the correct pattern in order to get a "resilient" session behavior in a scenario with underlying network connection breaking randomly and frequently?

Thanks.

milankase, 2016-12-09:
I believe you can use `SessionReconnectHandler` for this. Check the client sample application to see how to use it.

maiorfi, 2016-12-10:
I tried to take advantage of SessionReconnectHandler by mean of same pattern used in sample client, i.e. binding it to KeepAlive event handler (although it's not so clear whether I should activate reconnect handler whenever ServiceResult.IsGood(keepAliveEventArgs.Status) or on the contrary case), but I can not recover from, say, a cable disconnected on server side, since I always get a disposed TrasnsportChannel, as shown in:

`throw new ObjectDisposedException("TransportChannel is not available.");` at line 243 of ClientBase.cs

Any suggestion?

maiorfi, 2016-12-10:
I found following pattern seems working as expected:

`
private static void SessionOnKeepAlive(Session session, KeepAliveEventArgs keepAliveEventArgs)
        {
            Trace.TraceInformation($"SessionOnKeepAlive : {keepAliveEventArgs.CurrentState} {keepAliveEventArgs.Status}");

            if (!Object.ReferenceEquals(session, _session)) return;

            if (ServiceResult.IsBad(keepAliveEventArgs.Status))
            {
                if (_reconnectHandler == null)
                {
                    _reconnectHandler = new SessionReconnectHandler();
                    _reconnectHandler.BeginReconnect(_session, 10000, (sender, args) =>
                    {
                        Trace.TraceInformation("Reconnect complete.");

                        try
                        {
                            // ignore callbacks from discarded objects.
                            if (!Object.ReferenceEquals(sender, _reconnectHandler))
                            {
                                return;
                            }

                            _session = _reconnectHandler.Session;
                            _reconnectHandler.Dispose();
                            _reconnectHandler = null;
                        }
                        catch (Exception exception)
                        {
                            Trace.TraceWarning($"BeginReconnect callback exception : {exception}");
                        }
                    });
                }
            }

        }
`

Thanks!

closed 2016-12-10 by maiorfi