OPCF-Members/UA-.NET-Legacy #93: ClickOnce does not support the request execution level 'requireAdministrator'.

DerGroncki, 2017-01-12:
I am using Microsoft Visual Studio Community 2015 with Update 3. I followed the “Server Development Tutorial” from the “official site page” without problems. Now I want to deploy the server with ClickOnce Deployment and I get the following error:

ClickOnce does not support the request execution level 'requireAdministrator'.

I would be thankful if someone could point me to the right direction. How to continue from here? What is the right way to deploy the server? As I said I would like to uses ClickOnce.

Thank you.
Michael

deadmann, 2017-11-01:
I don't like to use any specific one, but yet i'm stuck too. i just want to simplify things when i turns so many DLL and files to my boss...

OneGenkiGirl73, 2020-05-19:
I'm stuck too, THIS IS A PROBLEM, because I'm trying to publish my solution for work ( and I don't seem to have any control over Microsoft automatically checking the "Enable ClickOnce Security Settings checkbox causing the error over and over no matter what I do...

I even moved the solution to VS 2019 and...walla...same problem... :-P 

**FIX IT !!!**

AlinMoldovean, 2020-05-19:
@OneGenkiGirl73 ,

Please check the _Readme_ section of this project there it states "_**Only critical security updates will be provided for this repository.**_"

https://github.com/OPCFoundation/UA-.NET-Legacy/blob/master/README.md