OPCF-Members/UA-.NET-Legacy #96: How do modify the address space after creation?

bowman-jm, 2017-01-18:
Hi! I am trying to follow the 'Server Development Tutorial' and have a few questions about some of the comments made in the guide:

"Keep in mind that the address space is created during application startup, but you can modify it at any time!" After start up, what class do I use to modify the address space an how exactly?

"You could define a single write function (event handler) that is used globally by all nodes within your application," Is there an example of this somewhere?

K47S, 2017-01-18:
If you are using a derived class from CustomNodeManager2:

1. Create your Node
2. Call the AddPredefinedNode(ISystemContext context, NodeState node) method

Example (maybe not correct, typed it from scratch):
```
var dataItemState = new DataItemState(parent);

            dataItemState.Create(
                systemContext.SystemContext,
                null,
                null,
                null,
                true);

nodeManager.AddPredefinedNode(null, dataItemState);
```

bowman-jm, 2017-01-19:
Thanks for the reply! 

I am working with the Quickstart solution and have the following code to start the OPC server:

            ApplicationInstance application = new ApplicationInstance();
            application.ApplicationType = ApplicationType.Server;
            application.ConfigSectionName = "Quickstarts.OPCServer";

            try
            {
                // load the application configuration.
                application.LoadApplicationConfiguration(false);

                // check the application certificate.
                application.CheckApplicationInstanceCertificate(false, 0);

                // start the server.
                var opcServer = new OPCServer();
                application.Start(opcServer);

                // Start the Message Loop for the processor
                Application.Run();

My NodeManager is instantiated during application.Start().. and once it starts I seemingly don't have access to it either through the ApplicationInstance or the OPCServer instance. Am I missing something or do I need to instantiate another NodeManager?

AlinMoldovean, 2017-05-03:
By default the NodeManager components are not available from StandardServer instances.
You need to expose new public members for this, e.g. set them in StandardServer.CreateMasterNodeManager

OpZed, 2017-06-07:
Could you expand further on this answer?  As far as I can tell all the examples have Statically defined Address Spaces.  Ideally it would be most helpful to have examples of both Server and Client which can Add and Delete Nodes from Form Events.

AlinMoldovean, 2017-06-09:
Hello @OpZed ,

The Server can create nodes in the address space during the runtime of the server process, by simply calling the AddPredefinedNode() method of the CustomNodeManager2 class.
An example can be found in BoilerNodeManager class of UA Sample Library project:
    SampleApplications\Samples\Common\Boiler\BoilerNodeManager.cs   
	private void CreateBoiler(SystemContext context, int unitNumber)
        {
            BoilerState boiler = new BoilerState(null);
            string name = Utils.Format("Boiler #{0}", unitNumber);

            boiler.Create(
                context,
                null, 
                new QualifiedName(name, m_namespaceIndex),
                null, 
                true);

            NodeState folder = (NodeState)FindPredefinedNode(
                ExpandedNodeId.ToNodeId(ObjectIds.Boilers, Server.NamespaceUris),
                typeof(NodeState));

            folder.AddReference(Opc.Ua.ReferenceTypeIds.Organizes, false, boiler.NodeId);
            boiler.AddReference(Opc.Ua.ReferenceTypeIds.Organizes, true, folder.NodeId);
            .....
            AddPredefinedNode(context, boiler);
        }
		
The CreateBoiler() method can be called during the runtime of the server application (instead of CreateAddressSpace method), e.g. at an application specific event callback.
The AddPredefinedNode will make the node(s) visible in the address space starting from that moment.

For adding nodes from the client side, the things are different.
The standard procedure in OPC UA requires the client to use the NodeManagement service set ( i.e. AddNodes, AddReferences, DeleteNodes, DeleteReferences) for this.
But for the moment there is no support and implementation for this set of services in the OPC Foundation UA .NET Stack.

A simple workaround can be implemented in the server application by exposing a Method node in the address space (e.g. CreateNode).
The server can create new nodes in the handle of the method call event.
Of course, the method will have some specific input arguments containing information for the node to be created.

closed 2017-06-19 by AlinMoldovean