OPCF-Members/UA-.NET-Legacy #126: Proper present Documentation of OPC-UA .NET SourceCode, Applications

sometody, 2017-06-13:
Does a proper present documentation of the OPC-UA .NET Sourcecode and Applications exist?
Free or to purchase??

Thanks 
Thomas

K47S, 2017-06-14:
Isn't this one suitable for you? [http://opcfoundation.github.io/UA-.NET/](http://opcfoundation.github.io/UA-.NET/)

sometody, 2017-06-14:
I hoped for more specific description of source code of the .NET applikations.

But thanks for your fast reply

kvloover, 2017-06-16:
Some parts are only sumarized briefly in the github.io documentation and often point to different code than alot of the samples.

For instance quickstarting your own server. This points to the reference server which uses a different base nodemanager than other samples.

MartMuell, 2017-09-13:
Download an old version of Netstack from OPC Foundation homepage and you will get a help-file for sourecode.