OPCF-Members/UA-.NET-Legacy #128: Receive Notification for not changed MonitoredItem

MRK89TP, 2017-06-14:
Hi, i'm working on schneider PacDriver3 PLC with OPC UA.
The client is on PC.
The server is on PLC.
I've a big number of variable in the PLC code.
The client create one session and one subscription, with one MonitoredItem for each variable on PLC.

If the variables on PLC code are declared "BOOL", everything works. The right notification is received by the client.
But if the variables are declared "BIT", for a single change the client receive 8 notifications, one of this for the real changed MonitoredItem, and others 7, for the MonitoredItems that are write near at the first item in the PLC code!

Example:

PLC                          PLC after change                      Notification on CLIENT

var...                        var....
var....                       var....
var1 false                var1 false                                    var1 true
var2 false                var2 false                                    var2 true
var3 false                var3 true                                     var3 true
var4 false                var4 false                                    var4 true
var5 false                var5 false                                    var5 true
var6 false                var6 false                                    var6 true
var7 false                var7 false                                    var7 true
var8 false                var8 false                                    var8 true
var...                        var...
var....                       var....

It seems a samplig problem.
There are any options that i can activate on client or on server, to fix this problem? 

Thanks.

AlinMoldovean, 2017-06-19:
Hello @MRK89TP,
The problem seems to be on the server side.
As you said, the server samples a byte value and updates the whole set of 8 UA variable bit nodes at any value change.

A fix from the client side could be to set the monitoring filter to "StatusValue"
A compliant server should should not generate data changes in case only the timestamp of the node was changed.

Something like this:
	// define a filter for the monitored item
	DataChangeFilter filter = new DataChangeFilter();
	filter.DeadbandType = (uint)DeadbandType.None;
	filter.Trigger = DataChangeTrigger.StatusValue;
	monitoredItem.Filter = filter;

MRK89TP, 2017-06-21:
Hi @AlinMoldovean ,
thanks for your interesting and answer.
I tried your solution, but do not works. I noticed that the serverTimeStamp and sourceTimeStampe are the same for all notifications.
I watched them as in Image.
[image]
I tried others filter trigger too, but do not works.

thanks

AlinMoldovean, 2017-06-21:
Hello @MRK89TP ,

In this case the problem is in the server and should be solved on that side.
It seems to have a non compliant behavior and generates unneeded datachange notifications.

closed 2017-09-28 by AlinMoldovean