OPCF-Members/UA-.NET-Legacy #141: Opc.Ua.ServiceResultException: The CertificateGenerator utility is not installed

alexanderastanin, 2017-08-04:
I develop a windows service that includes OPC UA server based on the Quickstarts framework. When I try to run my service then I get an exception "The CertificateGenerator utility is not installed". 

I have tried to run Opc.Ua.CertificateGenerator.exe manually via command line with "-cmd install" arguments, but I get an error "-error could not access certificate store:".  OPC UA Server generates only "\ProgramData\OpcFoundation\MachineDefault\" folder structure without any certificate file.

I have found a workaround to fix the problem - if I install OPC UA Sample Applications, then, after installation server works properly without any error. However, this installation includes sample apps aren't required for me. 

1) How can I install CertificateGenerator utility only (without sample applications)?
2) Any idea why I get an error "-error could not access certificate store:"? 
3) Where certificate generator utility should be placed to be able for my OPC UA Server?

astrowalker, 2017-08-04:
@alexanderastanin , put CG executable next to your executables and that's entire installation (of course maintaining certificates is another story). Maybe there are other ways, but this one works fine for me.

AlinMoldovean, 2017-08-04:
The CertificateGenerator utility is deployed by the Loca lDiscovery Server installation (included in OPC UA Sample Applications package)

closed 2017-09-28 by AlinMoldovean