OPCF-Members/UA-.NET-Legacy #147: NodeState.AddProperty() causes "Unexpected error creating address space"... am I doing this right?

josh2112, 2017-09-22:
Hi all... I'm trying to add a property to a FolderState:

```
var mwFolderVersion = microwaveFolderState.AddProperty<int>( "NodeVersion", DataTypeIds.Int32, ValueRanks.Scalar );
mwFolderVersion.Value = 1
```

When I go to create the address space via ```AddPredefinedNode()```, I get the following exception:

```
11100 - 9/22/2017 12:10:40.149 Unexpected error creating address space for NodeManager #3. ArgumentNullException 'Value cannot be null.
Parameter name: key'

========================================
Id: Bad
Description: Value cannot be null.
Parameter name: key
>>> Value cannot be null.
Parameter name: key
---    at Opc.Ua.NodeIdDictionary`1.set_Item(NodeId key, T value)
---    at Opc.Ua.Server.CustomNodeManager2.AddPredefinedNode(ISystemContext context, NodeState node)
---    at Opc.Ua.Server.CustomNodeManager2.AddPredefinedNode(ISystemContext context, NodeState node)
---    at Opc.Ua.Server.CustomNodeManager2.AddPredefinedNode(ISystemContext context, NodeState node)
---    at Opc.Ua.Server.CustomNodeManager2.AddPredefinedNode(ISystemContext context, NodeState node)
---    at CEM.LibertyBlueApplication.Libs.OPCServer.MethodRunnerNodeManager.CreateAddressSpace(IDictionary`2 externalReferences) in C:\CEM\LibertyBlueApplication\LibertyPro\LibertyProOPCServer\MethodRunnerNodeManager.cs:line 64
---    at Opc.Ua.Server.MasterNodeManager.Startup()
========================================

```

Even though that doesn't tell me which node is in error I know it's the property because the error goes away when I comment out those two lines.

This is part of a larger issue where I'm trying to notify clients of a new node being added to the model at runtime.  From what I've been able to cobble together, I need to do 2 things:

1.  Add a "NodeVersion" property to the folder which will accept new nodes.
2.  When I add a new node to the folder, somehow fire off a model-changed event... I know this has something to do with ```ModelChangeStructureDataType``` but I've not been able to find any documentation or sample code on this???

If these need to be separated into two issues I'll gladly do so.  Thanks!

fchaxel, 2018-03-09:
Hi @josh2112,

A long time without response indeed.  Maybe it's always required for you.

You have to call microwaveFolderState.Create ( ... ) after the new.

But this property is absolutely not required to force OpcUaExpert to rebrowse a folder  (I think it's what you want) after adding something in it.

Here assume RabbitRoot is the FolderState variable.
The following code is OK 

                GeneralModelChangeEventState ev = new GeneralModelChangeEventState(RabbitRoot);
                ev.Initialize(SystemContext,RabbitRoot,EventSeverity.Medium,new LocalizedText("Model change"));

                ev.Changes = new PropertyState<ModelChangeStructureDataType[]>(null);
                ev.Changes.Value = new ModelChangeStructureDataType[1];
                ev.Changes.Value[0] = new ModelChangeStructureDataType();
                ev.Changes.Value[0].Affected = RabbitRoot.NodeId;
                ev.Changes.Value[0].AffectedType = ObjectTypeIds.FolderType;
                // http://documentation.unified-automation.com/uasdkdotnet/2.5.2/html/classUnifiedAutomation_1_1UaBase_1_1ModelChangeStructureDataType.html
                ev.Changes.Value[0].Verb = 4; // 4 RefAdd, 8 RefDelete : 

                ev.SetChildValue(SystemContext, BrowseNames.SourceNode, RabbitRoot.NodeId, false);
                ev.SetChildValue(SystemContext, BrowseNames.SourceName, RabbitRoot.DisplayName, false);
                Server.ReportEvent(ev);

josh2112, 2018-03-09:
Thanks @fchaxel !  That is exactly what I want -- to notify the client to rebrowse a folder because I've added or removed a node.  I will give it a try later today and report back.

closed 2018-03-16 by josh2112