OPCF-Members/UA-.NET-Legacy #149: how to add nodes from the OPCUA Client side

gaoleiedison, 2017-09-29:
how to add nodes from the Client side. how to use the services as AddNodes,AddReferences

astrowalker, 2017-09-29:
Please take a look at the samples -- such basics are shown there.

gaoleiedison, 2017-09-29:
there are no sample for how to use addnodes

MartMuell, 2017-10-02:
``   For console application you can us 

  

```
>  var list = new List<MonitoredItem>(); 
>                list.Add(new MonitoredItem() {DisplayName = "ServerStatusCurrentTime", StartNodeId ="i=2258", AttributeId = Attributes.Value});
>                list.Add(new MonitoredItem() {DisplayName = "Counter1", StartNodeId = "ns=5;s=Counter1" , AttributeId = Attributes.Value });
>               //list.Add(new MonitoredItem() {}); // AttributeId, CacheQueueSize, DiscardOldest, DisplayName, Encoding, Filter, Handle, IndexRange
>                                                    // MonitoringMode, NodeClass, QueueSize, RelativePath, ResolvedNodeId, SamplingInterval, StartNodeId, Subscription
>                  list.ForEach(i => i.Notification += OnNotification); //Notification: wenn eine neue Benachrichtigung eintrifft
>                 
>                 subscription.AddItems(list);
```

AlinMoldovean, 2017-10-05:
Hello @gaoleiedison 
For the moment there is no default implementation for AddNodes/AddReferences services in the Server library, so you need to implement it in your application.
Another option can be to expose a method node in the server for this, e.g. an "AddNode" method with some input arguments like NodeId, NodeClass, BrowseName, DisplayName...

gaoleiedison, 2017-10-09:
So there's no way to use the services as AddNodes from Client side.

dathlin, 2017-10-11:
`public virtual ResponseHeader AddNodes(RequestHeader requestHeader, AddNodesItemCollection nodesToAdd, out AddNodesResultCollection results, out DiagnosticInfoCollection diagnosticInfos);`
I saw on SessionClient class , but server is not suppoted

AlinMoldovean, 2017-10-11:
The client has support for sending the AddNodes service request.
But you will need a server that has an implementation for it.

Something like this:

 string m_parentNodeId = "ns=0;i=85";
                    // create service request data
                    AddNodesItemCollection nodesToAdd = new AddNodesItemCollection();

                    // Create an Object node.
                    AddNodesItem node1 = new AddNodesItem();
                    node1.ParentNodeId = new NodeId(m_parentNodeId);
                    node1.ReferenceTypeId = ReferenceTypes.Organizes;
                    node1.RequestedNewNodeId = new ExpandedNodeId(1000,2);
                    node1.BrowseName = new QualifiedName("ObjectNode1");
                    node1.NodeClass = NodeClass.Object;
                    node1.NodeAttributes = null;
                    node1.TypeDefinition = ObjectTypeIds.FolderType;

                    //specify node attributes.
                    ObjectAttributes node1Attributes = new ObjectAttributes();
                    node1Attributes.DisplayName = "ObjectNode1";
                    node1Attributes.Description = "ObjectNode1 Description";
                    node1Attributes.EventNotifier = EventNotifiers.None;
                    node1Attributes.WriteMask = (uint)AttributeWriteMask.None;
                    node1Attributes.UserWriteMask = (uint)AttributeWriteMask.None;
                    node1Attributes.SpecifiedAttributes = (uint)(NodeAttributesMask.DisplayName | NodeAttributesMask.Description
                        | NodeAttributesMask.EventNotifier | NodeAttributesMask.WriteMask | NodeAttributesMask.UserWriteMask );

                    node1.NodeAttributes = new ExtensionObject(node1Attributes);
                    nodesToAdd.Add(node1);

                    // Create a Variable node.
                    AddNodesItem node2 = new AddNodesItem();
                    node2.ParentNodeId = new NodeId(m_parentNodeId);
                    node2.ReferenceTypeId = ReferenceTypes.HasComponent;
                    node2.RequestedNewNodeId = null;
                    node2.BrowseName = new QualifiedName("DataVariable1");
                    node2.NodeClass = NodeClass.Variable;
                    node2.NodeAttributes = null;
                    node2.TypeDefinition = VariableTypeIds.BaseDataVariableType;

                    //specify node attributes.
                    VariableAttributes node2Attribtues = new VariableAttributes();
                    node2Attribtues.DisplayName = "DataVariable1";
                    node2Attribtues.Description = "DataVariable1 Description";
                    node2Attribtues.Value = new Variant(123);
                    node2Attribtues.DataType = (uint)BuiltInType.Int32;
                    node2Attribtues.ValueRank = ValueRanks.Scalar;
                    node2Attribtues.ArrayDimensions = new UInt32Collection();
                    node2Attribtues.AccessLevel = AccessLevels.CurrentReadOrWrite;
                    node2Attribtues.UserAccessLevel = AccessLevels.CurrentReadOrWrite;
                    node2Attribtues.MinimumSamplingInterval = 0;
                    node2Attribtues.Historizing = false;
                    node2Attribtues.WriteMask = (uint)AttributeWriteMask.None;
                    node2Attribtues.UserWriteMask = (uint)AttributeWriteMask.None;
                    node2Attribtues.SpecifiedAttributes = (uint)NodeAttributesMask.All;

                    node2.NodeAttributes = new ExtensionObject(node2Attribtues);
                    nodesToAdd.Add(node2);
AddNodesResultCollection results = null;
                    DiagnosticInfoCollection diagnosticInfos;

                    RequestHeader requestHeader = new RequestHeader();
                    requestHeader.ReturnDiagnostics = (uint)DiagnosticsMasks.All;

m_session.AddNodes(
                        requestHeader,
                        nodesToAdd,
                        out results,
                        out diagnosticInfos);

gaoleiedison, 2017-10-11:
Thanks,may be i have to use the mothods provided from server side.

dathlin, 2017-10-12:
I have tried, but server returns something wrong, just like this:
BadServiceUnsupported
How to config server so that it supported ?
Thanks!

AlinMoldovean, 2017-10-12:
Hello @dathlin ,

The server component of this repository does not have implementation for the AddNodes service.
So is not a matter of setting some configuration parameters.

It requires some further development effort in your server application for implementing the service call handler.

closed 2017-10-12 by AlinMoldovean

dathlin, 2017-10-12:
Thank u very much , ^_^

YasminAkaichi, 2021-06-30:
Hello @AlinMoldovean, i wonder if there is a way for putting a standard value to  NodeAttribute.Value so it could accept both scalar and array types ?

AlinMoldovean, 2021-07-02:
Hello @YasminAkaichi ,

A Variable node can accept both scalar and array values if you set the _ValueRank_ attribute to _ScalarOrOneDimension (**-3**)_ or _Any (**-2**)_

YasminAkaichi, 2021-07-21:
Hello @AlinMoldovean thank you for your help. I have a problem with DataType. Is there a way that this field is initialized in a dynamic way ? I can't get to initialize it with the datatype of the item passed in parameter of the method. 
variable.DataType = item.NodeAttributes.Body : can't access  the datatype. What should I do ? (Variable 's type is a NodeVariable)