OPCF-Members/UA-.NET-Legacy #153: Images missing from ClientUtils().ImageList

SoftwareDave, 2017-11-13:
There is only one image defined in the ImageList control in ClientUtils.cs[Design].  And it's called TBD.
As such, when we use the BrowseTree control, and other list controls in the SampleClient (and other Clients), all the Treeview images are the same generic "TBD" icon.
Previous versions of the UA sample code had an ImageList that contained 35 images, including { Folder, Object, Property, Variable, Method, ObjectType, etc... }.

To reproduce:  
right click on ClientUtils.cs, click on View Designer.
right click on the ImageList control just below the designer pane.
select Choose Images, to bring up the Image Selector Dialog.
Note that there is only one image:  the TBD image.

My guess at a workaround:
I could copy the 35 images from the 2010 code base into the empty Images folder of the 2017 code base, and add them manually to the ImageList control, in the correct order.

My system:
VS 2017 Community version 15.4.3
targeting .NET 4.6.1.  
my computer: Windows 7 Ultimate

AlinMoldovean, 2017-11-14:
Hell @SoftwareDave ,

As far as I know, those images were removed due to unknown copyright source(s).
Being just a sample code,that control library is intended to be just a functional example and it should be extended with a proper user interface.

closed 2017-11-14 by AlinMoldovean

SoftwareDave, 2017-11-14:
Thanks - it's interesting, however, that there is an "icons" folder that has almost all of the missing BMPs.  Perhaps these are the non-copyrighted images?

.\UaNetMaster\SampleApplications\SampleLibraries\Controls\**Icons**

Array.bmp
Attribute.bmp
ByteString.bmp
ClosedFolder.bmp
Constant.bmp
InputArgument.bmp
Method.bmp
Number.bmp
Object.bmp
Object2.bmp
ObjectType.bmp
OpenFolder.bmp
OutputArgument.bmp
Property.bmp
Reference.bmp
String.bmp
Structure.bmp
Value.bmp
Variable.bmp
Varible.bmp
View.bmp