OPCF-Members/UA-.NET-Legacy #167: MatrikonOPC Explorer doesn't see sample application server

Talh4, 2018-01-25:
We are running the client and server applications on the sample application and we can connect from client to server but we can't connect from MatrikonOPC  Server Explorer to server or any other OPC Explorer's. Can you help us why is this happening ?

AlinMoldovean, 2018-01-25:
Hello @Talh4 ,

Could you check if the trust relationship between application certificates is configured?

Talh4, 2018-01-25:
I will check it first thing in the morning when i get to work and let you know. We are very noob at OPC connections so thank you for trying to help.

Talh4, 2018-01-26:
@fchaxel Oh now i see. Thank you for information. So i should find a DA based server and try it with MatrikonOPC then.