OPCF-Members/UA-.NET-Legacy #170: Release 1.04

benjamin1988, 2018-02-07:
Until when is it planned to provide the new types of the OPC UA Release 1.04?

dk2129, 2018-02-07:
The main focus of the OPC Foundation is "UA-.NETStandard".
The new types and NodeIds will be integrated within the next month or so. People willing to continue with "UA-.NET-Legacy" will have to take the files from "UA-.NETStandard".

AlinMoldovean, 2018-02-07:
The latest nodeset files and generated code are also available on the "UA-Nodeset" repository here:
   https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04

OPC Founadtion will soon update the "UA-.NETStandard" repository once all 1.03 related stuff will be done.
This repository will not receive V 1.04 types and feature implementations.

Thanks

closed 2018-02-07 by AlinMoldovean

SoftwareDave, 2018-02-07:
That's a problem:  the UA-.NETStandard requires several NuGet packages to be installed, and that's installed per application, not just a one time install as DLLs or libraries.  That's over 650MB of extra overhead per application.

**Question:**  Is there a way to install and run the UA-.NETStandard package and compile/build the DLLs and EXEs without the cumbersome NuGet packages and dependencies?  That's why we chose to go with the UA-.NET-master.zip (here after now called the UA-.NET-Legacy-master.zip) - it has no NuGet dependencies.