OPCF-Members/UA-.NET-Legacy #171: "Invalid provider type specified" loading certificate?

josh2112, 2018-03-15:
I've had a working OPC server integrated into my product for several months.  I don't know much about certificates.  I recently had an issue with my Windows 10 PC so I performed a "reset".  After reinstalling Visual Studio and rebuilding and running my server, I get this exception inside Opc.Ua.Configuration.ApplicationInstance.Start():

```
   at System.Security.Cryptography.CryptographicException.ThrowCryptographicException(Int32 hr)
   at System.Security.Cryptography.X509Certificates.X509Utils._LoadCertFromBlob(Byte[] rawData, IntPtr password, UInt32 dwFlags, Boolean persistKeySet, SafeCertContextHandle& pCertCtx)
   at System.Security.Cryptography.X509Certificates.X509Certificate.LoadCertificateFromBlob(Byte[] rawData, Object password, X509KeyStorageFlags keyStorageFlags)
   at System.Security.Cryptography.X509Certificates.X509Certificate2..ctor(Byte[] rawData, String password, X509KeyStorageFlags keyStorageFlags)
   at Opc.Ua.CertificateFactory.Load(X509Certificate2 certificate, Boolean ensurePrivateKeyAccessible)
   at Opc.Ua.CertificateIdentifier.Find(Boolean needPrivateKey)
   at Opc.Ua.ServerBase.OnServerStarting(ApplicationConfiguration configuration)
   at Opc.Ua.Server.StandardServer.OnServerStarting(ApplicationConfiguration configuration)
   at Opc.Ua.ServerBase.Start(ApplicationConfiguration configuration)
   at Opc.Ua.Configuration.ApplicationInstance.Start(ServerBase server)
   at CEM.LibertyBlueApplication.Libs.OPCServer.OPCServer.Start() in C:\CEM\LibertyBlueApplication\LibertyPro\LibertyProOPCServer\OPCServer.cs:line 79
   at CEM.LibertyBlueApplication.MainWindow.<ConfigureOPCServer>b__50_0() in C:\CEM\LibertyBlueApplication\LibertyPro\LibertyPro\MainWindow.xaml.cs:line 642
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
```

It seems to be having a problem with the certificate -- it creates it, then immediately fails trying to load it.  Is "Invalid provider type specified" a clue??

```
PID:4364 ************************* Logging started at 03/15/2018 16:17:08 *************************
4364 - 3/15/2018 16:17:08.756 Checking application instance certificate.
4364 - 3/15/2018 16:17:08.774 Creating application instance certificate. KeySize=0, Lifetime=600
4364 - 3/15/2018 16:17:08.780 Deleting application instance certificate.
4364 - 3/15/2018 16:17:11.111 Adding certificate to trusted peer store. StorePath=%CommonApplicationData%\CEM Corporation\Liberty PRO\CertificateStores\Trusted Applications
4364 - 3/15/2018 16:17:11.122 Certificate created. Thumbprint=2FF19ABB1A57B132C3E3295C9BC2197298F7DEDD
4364 - 3/15/2018 16:17:11.184 Could not load private key certificate from file: Liberty PRO OPC Server [2FF19ABB1A57B132C3E3295C9BC2197298F7DEDD].der CryptographicException 'Invalid provider type specified.
'

========================================
Id: Bad
Description: Invalid provider type specified.

>>> Invalid provider type specified.

---    at System.Security.Cryptography.Utils.CreateProvHandle(CspParameters parameters, Boolean randomKeyContainer)
---    at System.Security.Cryptography.Utils.GetKeyPairHelper(CspAlgorithmType keyType, CspParameters parameters, Boolean randomKeyContainer, Int32 dwKeySize, SafeProvHandle& safeProvHandle, SafeKeyHandle& safeKeyHandle)
---    at System.Security.Cryptography.RSACryptoServiceProvider.GetKeyPair()
---    at System.Security.Cryptography.RSACryptoServiceProvider..ctor(Int32 dwKeySize, CspParameters parameters, Boolean useDefaultKeySize)
---    at System.Security.Cryptography.X509Certificates.X509Certificate2.get_PrivateKey()
---    at Opc.Ua.DirectoryCertificateStore.LoadPrivateKey(String thumbprint, String subjectName, SecureString password)
========================================
```

astrowalker, 2018-03-16:
I don't know anything, but the error says "Liberty PRO __OPC__ Server" -- OPC is a different thing than OPC-UA. You don't build OPC server using OPC UA stack, do you?

josh2112, 2018-03-16:
I did; I'm in the right place.  My code is built with UA-.NET and based on the UA Sample Server project.

astrowalker, 2018-03-16:
AFAIK it won't work -- if you want to develop OPC soft use OPC stack, if you want to develop OPC-UA soft use OPC-UA stack, no mixing.

josh2112, 2018-03-16:
@astrowalker , I think you're getting hung up on terminology and missing my actual question. As I said, my code is based on the UA Sample Server found in the UA-.NET (now UA-.NET Legacy) sample code.  If you look at the stack trace and the log provided in the first post you can see the framework used.  If there's an issue with me referring to it as OPC instead of OPC-UA, I apologize, maybe a poor naming choice on my part; but that's not relevant to question.

As stated in my original post, I have had this server working for some time now.  I connect to it frequently from the UA Expert client for testing. It's worked fine on every PC up to this point.  My question is not one of development or choice of framework.

Does anyone else have any insight?  I have seen posts such as [this one from StackOverflow](https://stackoverflow.com/questions/22581811/invalid-provider-type-specified-cryptographicexception-when-trying-to-load-pri) suggesting that on my particular system the certificate is getting read into the wrong "provider", but I'm in over my head on certificate issues like this.