OPCF-Members/UA-.NET-Legacy #178: How to use UA Sample Client to connect siemens pcu 4.07 ua server with account and password?

acer1204, 2018-09-07:
I used UA sample Client to connect siemens pcu 4.05 ua server. is working(Anonymous Mode).

but when i use UA sample Client to siemens  pcu 4.07 ua server. it show bad certificate.(4.07 have to input account and password)

I have entered the correct account and password. but it show bad certificate.

when i use third-party software "softing opc ua client", and entered the correct account and password. it working.

how can i use UA sample Client to connect siemens  pcu 4.07 ua server?

fchaxel, 2018-10-17:
UA.NET Legacy is not the actual C# stack (dead), have a look to UA.NET Standard ... that's a reason why you did not get any response for a so long time.

For Siemens OPC, a lot of products are using SHA1 as a certificate hach algorithm. The UA.NET Standard do not accept this kind of certificate by default. Don't know with the old UA.NET Legacy.

If it's the same, on the code have a look to the SecurityConfiguration property of the ApplicationConfiguration object then put the RejectSHA1SignedCertificates property to false. 

Something like this
config.SecurityConfiguration.RejectSHA1SignedCertificates = false; // for Siemens PLC

or have a look to Opc.Ua.SampleClient.Config.xml if used by your code to add it inside.

Nerslob, 2025-05-21:
One question,
how did you get the server to run on a 4.05, i followed the instructions but the server is not reachable.

[https://kb.factorywiz.com/portal/en/kb/articles/add-machine-siemens-opc-ua-v4-05](url)