OPCF-Members/UA-.NET-Legacy #179: opc ua sdk got crashed while doing  uastring private class

govindsajgane, 2018-09-10:


closed 2018-09-11 by AlinMoldovean

govindsajgane, 2018-09-11:
In which version issue fixed. I need to discuss regarding issue with you .can you pleaase help for the same as its from opcua sdk side.

govindsajgane, 2018-09-11:
I need help regarding the same issue.

dk2129, 2018-09-12:
The ReadMe (https://github.com/OPCFoundation/UA-.NET-Legacy) states:
_This repository is provided by OPC Foundation as legacy support for older .NET Framework versions and will not receive further features and updates.
The official Unified Architecture .NET Stack maintained by OPC Foundation will be based on .NET Standard (https://github.com/OPCFoundation/UA-.NETStandard).
Only critical security updates will be provided for this repository._

Fixes provided by the community via Pull requests are welcome.