OPCF-Members/UA-.NET-Legacy #181: The first namespace in the table must be the OPC-UA namespace.

VinLexn, 2018-11-05:


ZbynekZ, 2018-11-05:
What you wrote is given by the OPC UA spec. But it is not clear why you are turning it into an issue on GitHub. Please describe where in the code, or under which circumstances the first namespace in the table isn't the OPC-UA namespace.

VinLexn, 2018-11-07:
When you connect to King IOServer with OPC UA client code, what is the reason for the return namespace list?
http://opcfoundation/UAProfile/Server/StandardUA
http://opcfoundation/UAProfile/Server/DataAccess
http://opcfoundation/UAProfile/Server/Methods
http://opcfoundation/UAProfile/Server/NodeManagement
http://opcfoundation/UAProfile/Server/EventSubscription
Under what circumstances does OPC UA access the returned namespace

ZbynekZ, 2018-11-07:
Namespace URI is part of the node or qualified name identification. Namespace index is just a means for efficient transport, so that the long namespaces URIs do not have to be transferred over the wire over and over. So, except for standardized namespace 0, any client should only know the namespace URIs, read in the namespace table, and convert the namespace URIs it needs into namespace indexes. This is also necessary because the server can rearrange the table and change the namespace indexes between sessions.

VinLexn, 2018-11-08:
Are namespaces generated by server-side software?Is namespace mismatch caused by older versions of server-side software?