OPCF-Members/UA-.NET-Legacy #184: ApplicationConfiguration.SaveToFile() removes "RejectSHA1SignedCertificates"

AntoniPancras, 2019-06-06:
Hi Team,

We are using Opc.Ua.Core.dll and Opc.Ua.Client.dll ( Version 1.4.354 )  in our application . Our application is targeted to .NET framework 4.6. Basically we have upgraded the OPC foundation SDK from Version 1.3.342  to version 1.04.354. 

When we try to communicate our OPC UA client application to any OPC UA server with any security mode ( Basic128Rsa15 , Basic 256 ) We are getting the  error message 
“SHA1 signed certificates are not trusted” .

It works fine with NONE security .

based on the reference https://github.com/OPCFoundation/UA-.NETStandard/issues/306, 
We have added below nodes under  ‘SecurityConfiguration’ in   our configuration file
  <RejectSHA1SignedCertificates>false</RejectSHA1SignedCertificates>
  <MinimumCertificateKeySize>1024</MinimumCertificateKeySize>

But the ApplicationConfiguration.SaveToFile() function removes the node "RejectSHA1SignedCertificates" from the configuration file .
Could you please advice us any solution for this?

 

Thanks
Antoni

AntoniPancras, 2019-06-10:
Hi Team,
We have debugged the source code of Opc.Ua.Core.dll and identified that the attribute "EmitDefaultValue" is set to 'false' for the property RejectSHA1SignedCertificates ( \Opc.Ua.Core\Schema\ApplicationConfiguration.cs. ) 

We have changed the attribute 'EmitDefaultValue' to 'true' as shown below 

  [DataMember(IsRequired = false, EmitDefaultValue = true, Order = 10)]

Now the issue is resolved. We are able to communicate with OPC server with any security policy successfully.

Please let us know your comments on this fix . 
Please do let us know any additional testing needs to be done.

Thanks
Antoni

PrashantDesai, 2020-03-26:
Is this problem is resolved in following Nuget package?? It is not working for me
<PackageReference Include="OPCFoundation.NetStandard.Opc.Ua">
      <Version>1.4.360.33</Version>
    </PackageReference>