OPCF-Members/UA-.NET-Legacy #186: EXECPTION: The UA COM Host process is not registered on the machine.

mary7091, 2019-10-09:
Hi,
in the UA Configuration Tool solution, I run the application and select manage COM interop, but when I select wrap UA server and select ua server address and other configuration, the exception with the message The UA COM Host process is not registered on the machine has thrown, what should I do? I test the server with UAclient and everything is ok

SoftwareDave, 2019-10-10:
This is just a guess:  it could have something to do with the Certifications on the client and server...  I know that sometimes I can get the UA supplied example Client(s) working, and yet my  customized Client throws an exception... and it all has to do with how the Client/Server certifications are handled.  This is not proposed as an exact solution, just a suggestion to try looking at the certs with the UA Config Tool, and if necessary, make new certs  so your Client is accepted by the server...  hope this helps in some way.

mary7091, 2019-10-12:
I tested your suggestion, but I got the same exception, in debug mode when I want to wrap UA server it checks for registry key in configUtils.cs
`RegistryKey key = Registry.ClassesRoot.OpenSubKey(String.Format(@"CLSID\{{{0}}}\LocalServer32", clsid));
`
and when it can't find the key, it returns null and throws an exception

SoftwareDave, 2019-10-13:
Once again, I'm just speculating:  check the Certificate Authority (Issuer Name) when you create the Certs with the Config tool.  We found that many time we had accidentally used the OPCUa Certificate Authority (UA Configuration Tool), when we should have been using our Organization's Certificate Authority.

WillemP-2, 2020-08-27:
I have the same issue. A manual exchange of certificates did not change it. I tries to find the clsid in registry, but this is not found, and the exception is raised.
Question is where it fails to do the registration of clsid?