OPCF-Members/UA-.NET-Legacy #192: browse more than 20k tags takes long using UA COM Server Wrapper application

isaacrimi, 2020-07-21:
Hello, 
I came across the following issue : I ran the UA Com Server Wrapper Sample and created a wrapper using a com server that has more than 20k tags under one branch. When I connected to the wrapper with an OPC UA Client and tried to browse that branch, the request generally times out. When it doesn’t, it took at least 2 minutes to complete and the branch is shown as empty. 
Why does it take that much time to browse?  Is there a way to optimize it?

SoftwareDave, 2020-07-21:
I've noticed it's slow too - I haven't had time yet to figure out where
that comes from - but to get around it, we browse the entire model when our
client app loads, so it looks like it's part of initialization - from then
on, it's not too bad.  Not a great workaround...

isaacrimi, 2020-07-27:
Hi, 
Thanks for the reply, I wanna ask how did you manage to browse the entire model?

SoftwareDave, 2020-07-29:
Since the model is a tree-like object, with parent nodes and child
nodes...  recursion is used.  at least that's the way I did it.
Every time you get a new node, you have to get all the children...  the
method calls itself (recursion).

Starting with the parent node root object,  and for namespaces > 1 (I don't
want the UA nodes, I want me model's nodes, which are namespaces > 1)
I used something similar to:
    Node childNode = Session.NodeCache.Find(expandedNodeId) as Node;
and if it meets my criteria, I add this childNode to an ongoing List:
    foundNodes.Add(childNode);

This childNode may have children of its own...  if the childNode is null
(meaning this parent has no children),
then the recursive method returns...

What's also good about this (besides getting the model into your own list),
it also puts it in the OPC/UA library's NodeCache.
Every search for a node after this is fast fast fast...

One thing I've learned in a lifetime of software - I count the recursion
layers - I used a static int to count how "deep" it goes.
On a normal 21st century PC or laptop, it's not a problem.

The NetMaster utilities, one of those many sample clients, probably has an
example of how to recursively get a set of all child nodes for a given
parent node.

If recursion sounds strange, complicated, new, etc. to you - there are
probably lots of simple recursive C# method examples on the interweb.
Take a look at those, and likely the pieces of the puzzle will fall into
place.

I hope this helps...
-Dave

isaacrimi, 2020-08-07:
Can you please just show a sniffet of how to browse recursively? how could i store the browsed items?