OPCF-Members/UA-.NET-Legacy #193: AE Addresspace in UA Wrapper is duplicated

isaacrimi, 2021-04-14:
Hi , 

I have noticied that in the createaddrespace in the ComAEClientNodeManager we link root to both Objects Folder And under Server Folder? Does removing the link to Sever Folder ( since it is duplicated ) have any effect? [
[image]
](url)

AlinMoldovean, 2021-04-15:
Hi @isaacrimi ,

The _HasNotifier_ reference is needed for reporting underlying Events on Server node.

isaacrimi, 2021-04-15:
Hi Alin, 

Yes, but why do link the root to both ObjectIds.Objects and ObjectIds.Server?

AlinMoldovean, 2021-04-15:
@isaacrimi ,

The Reference for Objects folder is the hierarchical reference that shows the underlying servers.
You can remove it consider that creates an overhead.